Nic30/hdlConvertor #146: VHDL: cannot parse for loop within a function

andrasm62, 2021-01-18:
Hello!

I tried to parse a VHDL file that contains some functions each with a for loop inside.

My function looks like this:
```vhdl
function F_FunctionName(
	v_in		: std_logic_vector)
	return std_logic
is
	variable tmp	: std_logic;
begin
	tmp := '0';
	A: for i in 0 to Ge_Item_Cnt - 1 loop
		if v_in(((i + 1) * 3) - 1 downto i * 3) = C_Value then
			tmp := '1';
		end if;
	end loop;
	return tmp;
end function;
```

The parses tries to find a FOR ... GENERATE loop, but it throws an exception (`raise TypeError("does not support HdlStmForIn", self, o)`), because the HdlStmForIn object's in_preproc attribute's value is False.

Nic30, 2021-01-21:
@andrasm62 it should be fixed you need to update also hdlConvertorAst, please confirm that the fix works for you so I can update pip packages as well.

https://github.com/Nic30/hdlConvertorAst/commit/8607949a8b420df0be0d85313908dafdeab29bc1

andrasm62, 2021-01-22:
Thank you, @Nic30 !
It seems to work, I could parse and generate an output from it which is the same (except the comments and whitespaces) as the input.

By the way, what do you think about adding to this new test, when there's a function definition that uses a FOR...LOOP?
I guess, the code won't make any difference that where you place this expression, but who knows, maybe later once it will be usefull.

Nic30, 2021-01-22:
> except the comments and whitespaces
* whitespaces are expected to differ as the code is completely newly formatted
* comments are missing because there is missing a function call which parses the comments.
   * it is easy to fix but I am quite short on time right now (I can provide an advice if you can fix it.) 

> what do you think about adding to this new test

* https://github.com/Nic30/hdlConvertor/commit/fd1a7c1342bad3c29d2480574e860437d1c869f3

It is a least I can do right now as it is easy to write test and it check that specified functionality.

The for loop can appear also on other places, some places do corresponds to a different rules in VHDL grammar. It would be better to walk through the grammar and check if the all unique appearances are tested.

andrasm62, 2021-01-22:
> it is easy to fix but I am quite short on time right now (I can provide an advice if you can fix it.)
I'm glad to help if it's easy to fix.

Nic30, 2021-01-22:
> I'm glad to help if it's easy to fix.

If you can provide some examples of comments for the tests it would be nice #148

I am closing this issue as you confirmed that the fix is working.

closed 2021-01-22 by Nic30