Nic30/hdlConvertor #147: VHDL: cannot visit NULL statement

andrasm62, 2021-01-22:
Hello!

I tried to parse a VHDL file that contains a CASE statement with a NULL statement.
An example from VHDL Reference Guide:
```vhdl
case ENCRYPTION is
  when "00" => 
    CPU_DATA_TMP := (B & A) - OPERAND;
  when "01" => 
    CPU_DATA_TMP := (B & A) + OPERAND;
  when "10" => 
    CPU_DATA_TMP := (A & B) - OPERAND; 
  when "11" => 
    CPU_DATA_TMP := (A & B) + OPERAND;
  when others =>
    null;
end case;
```

I think adding to hdlConvertorAst.to.common.py the `elif is None: w('NULL')` will solve the issue.

Nic30, 2021-01-22:
> I think adding to hdlConvertorAst.to.common.py.visit_iHdlExpr() the elif is None: w('NULL') will solve the issue.

Yes that was one of the possibilities however there is a HdlStmNop which should be used instead as it is HDL language independent.

andrasm62, 2021-01-25:
It seems to work! Thank you!

closed 2021-01-25 by Nic30