Nic30/hdlConvertor #149: SystemVerilog import package - doesn't seem to do anything?

flemish4, 2021-01-26:
Hi,
I'm trying to use hdlConvertor to discover all the files I need in my file list. (If someone has already done this please let me know :) ) 

So far I'm checking the error messages for missing includes directories and I'm looking for hdlAst._structural.HdlCompInst to get modules which are needed. Then I'm exploring the hierarchy and checking each module file found. This seems to work well.

The one thing that is missing is where there is an 'import something_pkg::*; this doesn't appear in the parsed object at all.
Then on lines where it is used there is just a double colon operator.

At no point does it realize that a file is missing.
Is this expected behaviour?

Thanks

flemish4, 2021-01-26:
Wow that was quick!
Thanks for the fix, I'll check it asap.

flemish4, 2021-01-27:
That looks to work well thanks, I can now extract the packages that are used.
It has revealed one more issue though - it seems that package declarations are not yet supported, is this planned?
"Source_textParser.visitDescription.package_declaration Conversion to Python object not implemented"

I have nested packages so this looks like the last hurdle to full file list generation :)

Thomasb81, 2021-01-27:
I guess contribution are welcome.

Basically : 
implement antlr visitePackage_declaration method and other visit* method relevant for your application in cpp.
Fill structure with data you want to collect and make it available into python world.
Implement unit test could help to preserve feature over tool change.

https://github.com/Nic30/hdlConvertor/commit/1cd4910ae1df8e15732760647f30797e87f9662e commit could be a useful source of inspiration.

flemish4, 2021-01-27:
> 1cd4910

Thanks for the pointers, I'll see what I can do

flemish4, 2021-01-27:
Is there a list of what is currently implemented vs not?

Thomasb81, 2021-01-27:
I don't think so.
That's why the tool can emit such message : "Source_textParser.visitDescription.package_declaration Conversion to Python object not implemented" at some place to point the limitation.

Nic30, 2021-01-27:
All things which are not converted to python yet are marked with  `NotImplementedLogger::print`
https://github.com/Nic30/hdlConvertor/blob/06fbd10227a0242cdd727f6cba3cc4aec8ee6f03/src/svConvertor/source_textParser.cpp#L92