Nic30/hdlConvertor #151: NotImplementedError: Unexpected object of type Error,

SingaDK, 2021-04-14:
Hi,
I am trying to parse Ibex RiscV core, files,  When I try to parse some files  and export to JSON, I get this error

NotImplementedError: Unexpected object of type <class 'hdlConvertorAst.hdlAst._expr.HdlExprNotImplemented'>

The error line is
```systemverilog
module ibex_alu #(
  parameter ibex_pkg::rv32b_e RV32B = ibex_pkg::RV32BNone
) (
    input  ibex_pkg::alu_op_e operator_i,
    input  logic [31:0]       operand_a_i,
    input  logic [31:0]       operand_b_i, ........)
```
"Ibex_pkg::rv32b" -> double colon.
I am using,

```python
d = c.parse(filenames, Language.SYSTEM_VERILOG_2017, include_dirs, hierarchyOnly=False, debug=True)
from hdlConvertorAst.to.json import ToJson
import json
tj = ToJson()
j = tj.visit_HdlContext(d)
# pretty print json
print(json.dumps(j, sort_keys=True,
                 indent=4, separators=(',', ': ')))
```
There is no error in parsing the file. The error happens only on `visit_HdlContext`
I can get JSON out of most files in the core, but not files with the above-mentioned parameter definition.
Any pointers on this will be highly helpful.
Many thanks in advance.

Nic30, 2021-04-15:
Hello,
can you try installation of latest version from git?
```
pip3 install git+git://github.com/Nic30/hdlConvertorAst.git@master#egg=hdlConvertorAst --upgrade --no-cache
pip3 install git+git://github.com/Nic30/hdlConvertor.git@master#egg=hdlConvertor --upgrade --no-cache
```

It should be fixed, but I can not immediately do a new release because there is some problem with documentation:
https://github.com/alendit/sphinx-pyreverse/issues/20#issuecomment-820214333

SingaDK, 2021-04-15:
Hi @Nic30,
Many thanks for the quick response and fix. 
I tried the new installation. I am not getting that error anymore instead, I got more. The parsing itself throws so many errors like
It mostly with `always ff`, `always comb`, `assertions`, `unique case`

> VerExprPrimaryParser.visitPrimaryAssig Conversion to Python object not implemented

> VerStatementParser.visitAlways_construct.always_keyword` - non pure always Conversion to Python object not implemented

> VerStatementParser.visitCase_statement.unique_priority`` Conversion to Python object not implemented

> VerGenerateParser.visitModule_or_generate_item.assertion_item Conversion to Python object not implemented

> VerModuleParser.visitModule_header_common - package_import_declaration Conversion to Python object not implemented

The file I am trying to parse is [ibex_alu.sv](https://github.com/lowRISC/ibex/blob/master/rtl/ibex_alu.sv).

Below I attached the full log of the error flow. Sorry for the long post, But many thanks in advance for looking into this.

```
/home/singa/ibex/rtl/ibex_alu.sv:51:2: VerStatementParser.visitAlways_construct.always_keyword - non pure always Conversion to Python object not implemented
    ...always_comb...
/home/singa/ibex/rtl/ibex_alu.sv:53:4: VerStatementParser.visitCase_statement.unique_priority Conversion to Python object not implemented
    ...uniquecase(operator_i)ALU_SUB,ALU_EQ,ALU_NE,ALU_GE,ALU_GEU,ALU_LT,ALU_LTU,ALU_SLT,ALU_SLTU,ALU_MIN,ALU_MINU,ALU_MAX,ALU_MAXU:adder_op_b_negate=1'b1;default:;endcase...
/home/singa/ibex/rtl/ibex_alu.sv:76:2: VerStatementParser.visitAlways_construct.always_keyword - non pure always Conversion to Python object not implemented
    ...always_comb...
/home/singa/ibex/rtl/ibex_alu.sv:77:4: VerStatementParser.visitCase_statement.unique_priority Conversion to Python object not implemented
    ...uniquecase(1'b1)multdiv_sel_i:adder_in_b=multdiv_operand_b_i;adder_op_b_negate:adder_in_b=operand_b_neg;default:adder_in_b={operand_b_i,1'b0};endcase...
/home/singa/ibex/rtl/ibex_alu.sv:99:2: VerStatementParser.visitAlways_construct.always_keyword - non pure always Conversion to Python object not implemented
    ...always_comb...
/home/singa/ibex/rtl/ibex_alu.sv:100:4: VerStatementParser.visitCase_statement.unique_priority Conversion to Python object not implemented
    ...uniquecase(operator_i)ALU_GE,ALU_LT,ALU_SLT,ALU_MIN,ALU_MAX:cmp_signed=1'b1;default:cmp_signed=1'b0;endcase...
/home/singa/ibex/rtl/ibex_alu.sv:116:2: VerStatementParser.visitAlways_construct.always_keyword - non pure always Conversion to Python object not implemented
    ...always_comb...
/home/singa/ibex/rtl/ibex_alu.sv:139:2: VerStatementParser.visitAlways_construct.always_keyword - non pure always Conversion to Python object not implemented
    ...always_comb...
/home/singa/ibex/rtl/ibex_alu.sv:140:4: VerStatementParser.visitCase_statement.unique_priority Conversion to Python object not implemented
    ...uniquecase(operator_i)ALU_EQ:cmp_result=is_equal;ALU_NE:cmp_result=~is_equal;ALU_GE,ALU_GEU,ALU_MAX,ALU_MAXU:cmp_result=is_greater_equal;ALU_LT,ALU_LTU,ALU_MIN,ALU_MINU,ALU_SLT,ALU_SLTU:cmp_result=~is_greater_equal;default:cmp_result=is_equal;endcase...
/home/singa/ibex/rtl/ibex_alu.sv:262:2: VerStatementParser.visitAlways_construct.always_keyword - non pure always Conversion to Python object not implemented
    ...always_comb...
/home/singa/ibex/rtl/ibex_alu.sv:284:2: VerStatementParser.visitAlways_construct.always_keyword - non pure always Conversion to Python object not implemented
    ...always_comb...
/home/singa/ibex/rtl/ibex_alu.sv:285:4: VerStatementParser.visitCase_statement.unique_priority Conversion to Python object not implemented
    ...uniquecase(operator_i)ALU_SLL:shift_left=1'b1;ALU_SLO,ALU_BFP:shift_left=(RV32B!=RV32BNone)?1'b1:1'b0;ALU_ROL:shift_left=(RV32B!=RV32BNone)?instr_first_cycle_i:0;ALU_ROR:shift_left=(RV32B!=RV32BNone)?~instr_first_cycle_i:0;ALU_FSL:shift_left=(RV32B!=RV32BNone)?(shift_amt[5]?~instr_first_cycle_i:instr_first_cycle_i):1'b0;ALU_FSR:shift_left=(RV32B!=RV32BNone)?(shift_amt[5]?instr_first_cycle_i:~instr_first_cycle_i):1'b0;default:shift_left=1'b0;endcase...
/home/singa/ibex/rtl/ibex_alu.sv:309:2: VerStatementParser.visitAlways_construct.always_keyword - non pure always Conversion to Python object not implemented
    ...always_comb...
/home/singa/ibex/rtl/ibex_alu.sv:315:6: VerStatementParser.visitCase_statement.unique_priority Conversion to Python object not implemented
    ...uniquecase(1'b1)bfp_op:shift_operand=bfp_mask_rev;shift_sbmode:shift_operand=32'h8000_0000;default:shift_operand=shift_left?operand_a_rev:operand_a_i;endcase...
/home/singa/ibex/rtl/ibex_alu.sv:351:2: VerStatementParser.visitAlways_construct.always_keyword - non pure always Conversion to Python object not implemented
    ...always_comb...
/home/singa/ibex/rtl/ibex_alu.sv:352:4: VerStatementParser.visitCase_statement.unique_priority Conversion to Python object not implemented
    ...uniquecase(operator_i)ALU_XNOR,ALU_ORN,ALU_ANDN:bwlogic_op_b_negate=(RV32B!=RV32BNone)?1'b1:1'b0;ALU_CMIX:bwlogic_op_b_negate=(RV32B!=RV32BNone)?~instr_first_cycle_i:1'b0;default:bwlogic_op_b_negate=1'b0;endcase...
/home/singa/ibex/rtl/ibex_alu.sv:371:2: VerStatementParser.visitAlways_construct.always_keyword - non pure always Conversion to Python object not implemented
    ...always_comb...
[... cut ...]
/home/singa/ibex/rtl/ibex_fetch_fifo.sv:381:2: VerGenerateParser.visitModule_or_generate_item.assertion_item Conversion to Python object not implemented
    ...IbexFetchFifoPushFull:assertproperty(@(posedgeclk_i)disableiff((!rst_ni)!=='0)((in_valid_i)|->(!valid_q[DEPTH-1]||clear_i)))elsebegin$error("%0t: (%0s:%0d) [%m] [ASSERT FAILED] %0s",$time,"/home/singa/ibex/rtl/ibex_fetch_fifo.sv",254,"IbexFetchFifoPushFull");end...

```
```

/home/singa/ibex/rtl/ibex_ex_block.sv:86:25: VerExprPrimaryParser.visitPrimaryAssig Conversion to Python object not implemented
    ...'{imd_val_q_i[0][31:0],imd_val_q_i[1][31:0]}...
```

Nic30, 2021-04-17:
`VerStatementParser.visitAlways_construct.always_keyword`
* fixed in f22f7e8d1f76e46382c07d671a40e90cf8686fc4

`VerStatementParser.visitCase_statement.unique_priority`
* fixed in 7db2e98754d7b4e9c052678649d12e0a4c2467f5

`VerExprPrimaryParser.visitPrimaryAssig`
* fixed in 39aa37382a5934554188c49fb8e3f5279848c8db

Now [ibex_alu.sv](https://github.com/lowRISC/ibex/blob/master/rtl/ibex_alu.sv) should be parsed successfuly , please check. If possible send pull request with minimal simple tests or at least provide minimal simple code examples which do not work or are not currently tested.

Nic30, 2021-05-14:
Fixed in mentioned commits.

closed 2021-05-14 by Nic30