Nic30/hdlConvertor #158: Shared Variables within VHDL not implemented?

hjnauman, 2021-05-25:
When parsing through a RISCV microprocessor which can be found here: https://github.com/f32c/f32c I noticed an issue with shared variables not being implemented yet from the file located at `rtl/generic/bram_true2p_1clk.vhd`? I think this warning is from running the visitor through the files `hdlContext` because from looking into the context itself it looks like the actual signal assignment for `ram` is represented as an `HdlIdDef` object. This is the warning that I got:
```text
/rtl/generic/bram_true2p_1clk.vhd:42:1: VariableParser.visitVariable_declaration - SHARED Conversion to Python object not implemented 
...sharedvariableram:memory_t;...
```

below is a snippet of the file in question that contains the shared variable declaration as well as some additional entity information:
```VHDL
library ieee;
use ieee.std_logic_1164.all;
use ieee.std_logic_unsigned.all;

entity bram_true2p_1clk is
	generic 
	(
		dual_port: boolean := True; -- set to False for single port A
		pass_thru_a, pass_thru_b: boolean := True;
		data_width: natural := 8;
		addr_width: natural := 6
	);
	port 
	(
		clk: in std_logic;
		addr_a: in std_logic_vector((addr_width-1) downto 0);
		addr_b: in std_logic_vector((addr_width-1) downto 0) := (others => '-');
		we_a: in std_logic := '0';
		we_b: in std_logic := '0';
		data_in_a: in std_logic_vector((data_width-1) downto 0);
		data_in_b: in std_logic_vector((data_width-1) downto 0) := (others => '-');
		data_out_a: out std_logic_vector((data_width -1) downto 0);
		data_out_b: out std_logic_vector((data_width -1) downto 0)
	);
end bram_true2p_1clk;

architecture rtl of bram_true2p_1clk is
	-- Build a 2-D array type for the RAM
	subtype word_t is std_logic_vector((data_width-1) downto 0);
	type memory_t is array(2**addr_width-1 downto 0) of word_t;

	-- Declare the RAM 
	shared variable ram: memory_t;
begin
```

Nic30, 2021-05-25:
Hello @hjnauman ,

the message originates from https://github.com/Nic30/hdlConvertor/blob/master/src/vhdlConvertor/variableParser.cpp#L18
the variable declaration should be present there is just not any flag that denotes that this variable was share.
The vhdl standard specifies when the variable must be shared (when written from multiple processes) in other cases the `shared` has no effect.

If you need it, it is easy to add.
Did you find out any case where you need to know if the variable is shared or not?

hjnauman, 2021-05-27:
At the moment I don't think I really need to know if the variable is shared or not, at least for my purposes, however, I was just thinking about it from a completeness standpoint for the parser.

Nic30, 2021-05-28:
ok, now it is implemented

hjnauman, 2021-05-28:
Awesome thank you!

closed 2021-05-28 by hjnauman

hjnauman, 2021-05-31:
Hey @Nic30, I just got around to updating my versions of `hdlConvertor` and `hdlConvertorAst` and I'm now getting this issue:
``` text
return c.parse(filenames, language, include_dirs, hierarchyOnly=hierarchyOnly, debug=debug)
  File "_hdlConvertor.pyx", line 178, in hdlConvertor._hdlConvertor.HdlConvertorPy.parse
AttributeError: 'HdlIdDef' object has no attribute 'is_shared'
```
the weird thing is my friend who has the same version of the library and is testing the exact same file isn't having this issue. I went as far as to uninstall python3 to try and get it working but even after reinstalling the issue persisted.

A small note that I feel should be made is that I'm running everything in WSL2 at the moment as it's easier to update the library. Not sure if this is affecting it at all but I'm not sure how to fix the issue.

hjnauman, 2021-05-31:
I'm not sure what was causing this issue, but I went ahead and completely reinstalled my linux distro and it did the trick