Nic30/hdlConvertor #166: incorrect Position

Thomasb81, 2022-02-05:
Hello

Position return by the tool is incorrect:
include.vh:
```
input a
```
test.v:
```

module my_module(
`include "include.vh"
);
endmodule
```
test.py
```
from hdlConvertorAst.language import Language
from hdlConvertor import HdlConvertor

c = HdlConvertor()
d = c.parse("test.v", Language.VERILOG, [])
print(d.objs[0])

```

Return
```
...
         'ports': [{'__class__': 'HdlIdDef',
                    'direction': 'IN',
                    'name': {'__class__': 'str', 'val': 'a'},
                    'position': (5, 1, 5, 7),
                    'type': 'wire'}],
...
```

5 refer to post processed source code which make no sens for an user.
We expect here an information as " include.vh, start_line 1, start_position 1,end_line 1, end_position 7"

Nic30, 2022-02-06:
Hello,

problem is that the line/file is not translated using [line_map](https://github.com/Nic30/hdlConvertor/blob/master/include/hdlConvertor/verilogPreproc/out_buffer.h#L32)

Thomasb81, 2022-02-06:
I am looking at that, but I am confuse because preprocessor result is different from what I would have expect.
include.vh
```
input a,
//`include "include2.vh"
output c
```
test.v
```

module my_module(
`include "include.vh"
);
endmodule
```

is translate into
```

module my_module(

input a,

//`include "include2.vh"

output c

);
endmodule
```
Newline have been inserted . 
The data are coherent but this is making thing difficult to understand. Especially if we want to extend the mechanic to provide to the parser the ability to take into account  some compîler directive `timescale` `default_nettype` `pragma` ... that can take different value in a source code and impact differently the model to be build.

Nic30, 2022-02-19:
Hello @Thomasb81 , have you had any success fixing this issue?
In next week I still do not have any time, but I think that this fix is something essential and simply must be done.
It is required to reuse the information about origin of the line, which should be already present.
It was left unfinished, because there was the question if we should change API to provide whole hierarchy of files and lines as they were added using include.

Thomasb81, 2022-02-19:
unfortunately no progress.

Nic30, 2022-06-16:
@Thomasb81 , theoretically I do have some people through summer jobs, I do not know if it is withing their reach but I hopefully the branch with the fix will be finally finished and merged.

Thomasb81, 2024-07-14:
Hello

The current status of the draft branch provide following result:
```
         'ports': [{'__class__': 'HdlIdDef',
                    'direction': 'IN',
                    'name': {'__class__': 'str', 'val': 'a'},
                    'position': (1, 1, 1, 7, 'include.vh'),
                    'type': 'wire'}],
```
Which look ok, but soon as we complexity a little the code, the position could become incorrect.
According to my analyses  data push in 
https://github.com/Nic30/hdlConvertor/blob/d623bc833e796480b243666064bbdee46bd2f027/include/hdlConvertor/verilogPreproc/file_line_map.h#L19

Lead to put wrong data on token. so probably need to revise how are collect and use the data.