Nic30/hdlConvertor #168: Convert VHDL package to Verilog or SV

jabate-anova, 2022-04-05:
I have VHDL package files in my current directory which I'd like to convert to Verilog or SV using hdlConvertor. Is this possible?

As in the [notebook example](https://github.com/Nic30/hdlConvertor/blob/master/notebooks/01_parse_and_dump.ipynb), I'm able to use ```visit_HdlContext(d)``` to display a VHDL package:
```
from hdlConvertor import HdlConvertor
from hdlConvertorAst.to.vhdl.vhdl2008 import ToVhdl2008

TEST_DIR = os.path.join("./")
filenames = [os.path.join(TEST_DIR, "axis_pkg.vhd"), ]
include_dirs = []
c = HdlConvertor()
d = c.parse(filenames, Language.VHDL, include_dirs, hierarchyOnly=False, debug=True)
to_vhdl = ToVhdl2008(sys.stdout) 
to_vhdl.visit_HdlContext(d) 
```

I see that the [hdlConvertorAst.translate](https://github.com/Nic30/hdlConvertorAst/tree/master/hdlConvertorAst/translate) module may be able to help me, but I'm not sure how to proceed. If I import VHDL to verilog (```from hdlConvertorAst.translate import vhdl_to_verilog```), how could I use ```vhdl_to_verilog.link_module_dec_def(context)``` with my parsed VHDL?

Any help is appreciated, thanks!

rubenm14, 2022-08-05:
Hi, Nic. 

I'm using hdlConvertor to parse a VHDL file to verilog. Following your recommendation, I'm using:

`from hdlConvertorAst.to.vhdl.vhdl2008 import ToVhdl2008`
`tv = ToVhdl2008(sys.stdout)`
`tvh.visit_HdlContext(d) `

to get the VHDL AST, but I I'm not following how to use hdlConvertorAst to verilog, do you have any tips/example?
Thanks for the help:)

poleguy, 2023-08-23:
This is the same question I was about to ask. I can't find any examples that produce valid verilog starting from even trivial VHDL. Can we produce a working python example from any of the 'cases' that are possible? Just as a starting point? And maybe a trivial working example that uses hdlConvertorAst.translat.vhdl_to_verilog?