Nic30/hdlConvertor #169: Verilog2017 - unpacked array converted as packed

diffore, 2022-06-14:
Hi, 

it seems that HdlConvertor does not preserve the dimensions of array when parsing the module interface. See the example below: 

**example.sv:** 
```verilog
module test (
logic [0:10] packed_array,   
logic unpacked_array[10:0] 
);
endmodule
```

**python code:**
```python     
c = HdlConvertor()
d = c.parse("example.sv", Language.SYSTEM_VERILOG_2017, include_dirs, hierarchyOnly=False)
tv = ToVerilog2005(sys.stdout)
tv.visit_HdlContext(d)
```

**Output (both arrays are packed now):**
```verilog
module test (
    inout logic[0:10] packed_array, 
    inout logic[10:0] unpacked_array
);
endmodule
```

Is this something that is not supported or a bug? 

P.S. I am trying to create a tool to generate interfaces from module descriptions and would like to preserve the exact port descriptions.

Nic30, 2022-06-16:
Hello @diffore,

>Is this something that is not supported or a bug?

 bug

this was fixed before, but it seems that the issue is still there, the problem is that originally arrays and bit vectors did use same notation and in some cases it was not possible to distinguish if it was vector or array.

I need to fix several other issues first, but I will fix this. Does this issue somehow blocks you in your work?

diffore, 2022-06-17:
Hey @Nic30, 

no, it does not block anything, so please take your time.
 
Thanks.