Nic30/hdlConvertor #170: KeyError when working with ternary operators and ranges?

shafe123, 2022-06-17:
Not sure if the summary is right as I'm not sure where to begin troubleshooting.

I've got the VHDL below from a library.  When parsing the VHDL and then attempting to write it back out to a file, I get an exception:  
`Exception has occurred: KeyError
<HdlOpType.MAP_ASSOCIATION: 81>
`

This is the Python that I'm using to parse/write the file.

`def writeToFile(vhdl, destination):
    with StringIO() as buff:
        serializer = ToVhdl2008(buff)
        serializer.visit_HdlContext(vhdl)

        with open(destination, "w") as of:
            result = buff.getvalue()
            of.write(result)

if __name__ == "__main__":
    convertor = HdlConvertor()
    file = "samples/vhdl/rightShifter_8.vhdl"
    vhdl = convertor.parse(file, Language.VHDL_2008, [])
    writeToFile(vhdl, "test/output.vhdl")`

VHDL below

`--------------------------------------------------------------------------------
--                    RightShifter_8_by_max_10_F700_uid80
-- This operator is part of the Infinite Virtual Library FloPoCoLib
-- All rights reserved 
-- Authors: Bogdan Pasca, Florent de Dinechin (2008-2011)
--------------------------------------------------------------------------------
-- Pipeline depth: 1 cycles

library ieee;
use ieee.std_logic_1164.all;
use ieee.std_logic_arith.all;
use ieee.std_logic_unsigned.all;
library std;
use std.textio.all;
library work;

entity RightShifter_8_by_max_10_F700_uid80 is
   port ( clk, rst : in std_logic;
          X : in  std_logic_vector(7 downto 0);
          S : in  std_logic_vector(3 downto 0);
          R : out  std_logic_vector(17 downto 0)   );
end entity;

architecture arch of RightShifter_8_by_max_10_F700_uid80 is
signal level0 :  std_logic_vector(7 downto 0);
signal ps, ps_d1 :  std_logic_vector(3 downto 0);
signal level1 :  std_logic_vector(8 downto 0);
signal level2, level2_d1 :  std_logic_vector(10 downto 0);
signal level3 :  std_logic_vector(14 downto 0);
signal level4 :  std_logic_vector(22 downto 0);
begin
   process(clk)
      begin
         if clk'event and clk = '1' then
            ps_d1 <=  ps;
            level2_d1 <=  level2;
         end if;
      end process;
   level0<= X;
   ps<= S;
   level1<=  (0 downto 0 => '0') & level0 when ps(0) = '1' else    level0 & (0 downto 0 => '0');
   level2<=  (1 downto 0 => '0') & level1 when ps(1) = '1' else    level1 & (1 downto 0 => '0');
   ----------------Synchro barrier, entering cycle 1----------------
   level3<=  (3 downto 0 => '0') & level2_d1 when ps_d1(2) = '1' else    level2_d1 & (3 downto 0 => '0');
   level4<=  (7 downto 0 => '0') & level3 when ps_d1(3) = '1' else    level3 & (7 downto 0 => '0');
   R <= level4(22 downto 5);
end architecture;`

Nic30, 2022-06-18:
Hello, can you please check that this is expected output so I can add it as a test?
```vhdl
LIBRARY ieee;
USE ieee.std_logic_1164.ALL;
USE ieee.std_logic_arith.ALL;
USE ieee.std_logic_unsigned.ALL;
LIBRARY std;
USE std.textio.ALL;
LIBRARY work;
ENTITY RightShifter_8_by_max_10_F700_uid80 IS
    PORT(
        clk : IN std_logic;
        rst : IN std_logic;
        X : IN std_logic_vector(7 DOWNTO 0);
        S : IN std_logic_vector(3 DOWNTO 0);
        R : OUT std_logic_vector(17 DOWNTO 0)
    );
END ENTITY;

ARCHITECTURE arch OF RightShifter_8_by_max_10_F700_uid80 IS
    SIGNAL level0 : std_logic_vector(7 DOWNTO 0);
    SIGNAL ps : std_logic_vector(3 DOWNTO 0);
    SIGNAL ps_d1 : std_logic_vector(3 DOWNTO 0);
    SIGNAL level1 : std_logic_vector(8 DOWNTO 0);
    SIGNAL level2 : std_logic_vector(10 DOWNTO 0);
    SIGNAL level2_d1 : std_logic_vector(10 DOWNTO 0);
    SIGNAL level3 : std_logic_vector(14 DOWNTO 0);
    SIGNAL level4 : std_logic_vector(22 DOWNTO 0);
BEGIN
    PROCESS(clk)
    BEGIN
        IF clk'event AND clk = '1' THEN
            ps_d1 <= ps;
            level2_d1 <= level2;
        END IF;
    END PROCESS;
    level0 <= X;
    ps <= S;
    level1 <= ((0 DOWNTO 0 => '0') & level0) WHEN (ps(0) = '1') ELSE level0 & (0 DOWNTO 0 => '0');
    level2 <= ((1 DOWNTO 0 => '0') & level1) WHEN (ps(1) = '1') ELSE level1 & (1 DOWNTO 0 => '0');
    level3 <= ((3 DOWNTO 0 => '0') & level2_d1) WHEN (ps_d1(2) = '1') ELSE level2_d1 & (3 DOWNTO 0 => '0');
    level4 <= ((7 DOWNTO 0 => '0') & level3) WHEN (ps_d1(3) = '1') ELSE level3 & (7 DOWNTO 0 => '0');
    R <= level4(22 DOWNTO 5);
END ARCHITECTURE;
```

The fix in is hdlConvertorAst, you can try it using
```
sudo pip3 install --upgrade --force-reinstall --no-cache-dir git+https://github.com/Nic30/hdlConvertorAst.git
```

shafe123, 2022-06-19:
That looks good at a first glance, let me double-check something and I'll get back to you with it.

shafe123, 2022-06-30:
This works perfectly, thanks @Nic30

closed 2022-06-30 by shafe123