Nic30/hdlConvertor #171: ANTLR 4.11.1 support

apatern0, 2022-10-06:
Hi!

I managed to compile hdlConvertor using ANTLR4 version 4.11.1, but I had to perform the following changes because of ANTLR4 switching to std::any.

Thought might be useful to share!

```
diff --git a/src/verilogPreproc/verilogPreproc.cpp b/src/verilogPreproc/verilogPreproc.cpp
index eab9f25..cba15b1 100644
--- a/src/verilogPreproc/verilogPreproc.cpp
+++ b/src/verilogPreproc/verilogPreproc.cpp
@@ -193,7 +193,7 @@ antlrcpp::Any VerilogPreproc::visitDefine(
        vector<MacroDefVerilog::param_info_t> *params;
        vector<MacroDefVerilog::param_info_t> params_dummy;
        if (da)
-               params = visitDefine_args(da);
+               params = std::any_cast<vector<MacroDefVerilog::param_info_t> *>(visitDefine_args(da));
        else {
                params = &params_dummy;
        }
@@ -403,7 +403,7 @@ antlrcpp::Any VerilogPreproc::visitInclude(
        string inc_file_path;
        auto mc = ctx->macro_call();
        if (mc) {
-               inc_file_path = visitMacro_call(mc, false).as<string>();
+               inc_file_path = std::any_cast<string>(visitMacro_call(mc, false));
        } else {
                inc_file_path = ctx->STR()->getText();
        }
```

Nic30, 2022-10-10:
Thank you for your contribution. However there is a problem. This is not compatible with 4.9 which is still default in Debian repositories.
Naturally we can just add ifdef to support both versions.

I did not find any def. of antrl4 in .h files of antlr. Do you know about any?
Otherwise it is required to check version in cmake and propagate it to cpp so we can add ifdef.

sfence, 2023-02-19:
On my system (Arch linux):
`#include <antlr4-runtime.h>` includes also /usr/include/antlr4-runtime/Version.h which contains macros  ANTLRCPP_VERSION_MAJOR, ANTLRCPP_VERSION_MINOR and ANTLRCPP_VERSION_PATCH.

So it can be useful for ifdef.

PR #174 (2023-04-08): Fix Antlr4.10 std::any usage
Fixes #171 .
closed 2023-04-08 by Nic30