Nic30/hdlConvertor #172: provided instructions don't work for finding antlr-complete.jar

bsferrazza, 2022-10-08:
Hi. I have antlr-4.11.1-complete.jar located at /home/myusername/tools/opt/jdk/lib/antlr-4.11.1-complete.jar.

I've tried setting this path using -DANTLR_JAR_LOCATION but that fails to work

```
CMake Error at src/CMake_antlr4.txt:21 (MESSAGE):
  ANTLR: Can not find stringtemplate4 jar from ANTLR installation
```

Looking into src/CMake_antlr4.txt it's unclear how I set what variable to actually get hdlConvertor to find this jar.

```
find_jar(_ANTLR_JAR_LOCATION_antlr4_complete "antlr4-complete"
        PATHS "/usr/share/java/antlr4-complete.jar"
                  "antlr4-complete.jar" ENV ANTLR_COMPLETE_PATH)
if(_ANTLR_JAR_LOCATION_antlr4_complete)
        MESSAGE(STATUS "using antlr4-complete.jar found in ${_ANTLR_JAR_LOCATION_antlr4_complete}")
        set(ANTLR_CLASSPATH "${_ANTLR_JAR_LOCATION_antlr4_complete}")
```

Please update the README.md to expand upon how to install hdlConvertor so that it finds antlr4 and all dependencies. As it is, there's no explanation on how to do so. Thank you!

bsferrazza, 2022-10-08:
Was able to get it to install, and figured I'd share my findings for a Linux build. I installed it as non-root, using a $PREFIX.

hdlConvertor requires antlr-4.x.x-complete.jar, but looks for antlr4-complete.jar. Where you have this jar installed, link antlr4-complete.jar to this file. For me that is `$PREFIX/opt/jdk/lib`

`ln -s antlr-4.1.1-complete.jar antlr4-complete.jar`

This is found with the `-DJava_JAR_PATHS` argument when installing hdlConvertor, by specifying the directory path where this jar is found.

You also need the antlr4-runtime libraries and header files. I built this from source by downloading and extracting [antlt4-cpp-runtime-source](https://www.antlr.org/download/antlr4-cpp-runtime-4.11.1-source.zip). You'll want the latest, which at the time of this post was 4.11.1.

when inside the directory where you extracted the antlr4-cpp-runtime source, issue the following to build and install. Adjust jar location and prefix according to your setup. /usr/local is the typical $PREFIX

```
mkdir build && cd build
cmake .. -DANTLR_JAR_LOCATION=$PREFIX/opt/jdk/lib/antlr-4.11.1-complete.jar -DCMAKE_INSTALL_PREFIX=$PREFIX -DWITH_DEMO=True
make
make install # use sudo if installing to a system folder such as /usr/local
```

Now you should be able to install hdlConvertor from source, as follows. This assumes you've cloned the git repository

`python3 setup.py install -- -DJava_JAR_PATHS=$PREFIX/opt/jdk/lib -DANTLR4CPP_ROOT=$PREFIX`

Nic30, 2022-10-10:
@bsferrazza I am glad you managed to install the library on your own. However I am disappointed in scikit-build and cmake, there is always some new issue.
I am planing o migrate project to meson and do all the required downloads there if required. But somehow I am constantly out of the time.

Do you think that issue is caused only by `-DCMAKE_INSTALL_PREFIX` or there is something tricky that needs to be in readme?
Are you building cpp based cmake managed build intentionally or you was originally intended to build a python library?

bsferrazza, 2022-10-11:
> @bsferrazza I am glad you managed to install the library on your own. However I am disappointed in scikit-build and cmake, there is always some new issue. I am planing o migrate project to meson and [...]

I'm using it simply to build the antlr4 shared object libraries and header files. I have built an entire bootstrapped toolchain with dozens and dozens of libraries and binaries that is insulated from the host system's libraries, hence the PREFIX. It's largely done at work where I don't have root access and want to have a fully insulated toolchain and set of binaries at my disposal.

closed 2022-10-11 by bsferrazza

Nic30, 2022-10-11:
@bsferrazza 

> I have built an entire bootstrapped toolchain

Is it an opensource? I was looking for some people to build "sv/vhdl frontend for llvm" but I never actually found anyone who had time for this. That is why I am curious.

bsferrazza, 2022-10-13:
@Nic30 it's a glibc+binutils+gcc toolchain, compiled from source. that I've then used to build tons of libraries and binaries, including other compilers and programming tools like LLVM clang, rustc, go, python, ruby, etc.

Nic30, 2022-10-13:
Sounds to me like some fakeroot/yocto/buildroot project. Anyway, if there is anything specific how can make this library more easy to use feel free to open another issue.