Nic30/hdlConvertor #182: CMake install DIRECTORY include wrong parameters?

liuzikai, 2023-06-13:
Hi,

https://github.com/Nic30/hdlConvertor/blob/56b2da68d494f8cec3e22d74e219415c151259f9/src/CMakeLists.txt#L96

According to [CMake doc](https://cmake.org/cmake/help/latest/command/install.html#installing-directories), there should be only one parameter following `DESTINATION`? When building as a standalone project, `${CMAKE_INSTALL_INCLUDEDIR}` is empty so nothing happens. But when I include this project in another one via `add_subdirectory` with this variable set (also `include` though), it breaks CMake.

Nic30, 2023-06-21:
@liuzikai Do you propose any solution to this problem?
There is a branch where I replaced cmake with meson. https://github.com/Nic30/hdlConvertor/tree/mesonbuild
But I am fighting against the time there...

liuzikai, 2023-06-21:
I am not sure either... A possible solution is to remove `${CMAKE_INSTALL_INCLUDEDIR}` or default it to "include" if it's empty like the following. Just make sure it doesn't break your python build (which I am not really familiar).
```
if(CMAKE_INSTALL_INCLUDEDIR STREQUAL "")
    set(CMAKE_INSTALL_INCLUDEDIR "include")
endif()
```

I was trying to use this library in another pretty large CMake-based project, whose configuration does some wired things to these variables. Simply remove either `include` or `${CMAKE_INSTALL_INCLUDEDIR}` works, although later I encountered some other problems and not using this lib for now. So anyway, feel free to apply these easy fixes or just close this issue as you like ;)

Nic30, 2025-06-10:
cmake replaced by meson, closing this issue