Nic30/hdlConvertor #183: Can't build master on ubuntu 22.04

poleguy, 2023-08-15:
Well, it builds, but it gives an error when you try to import it. Here's the output. I'm going to try building older releases to see if anything will work.

python
Python 3.7.16 (default, Jan 17 2023, 22:20:44) 
[GCC 11.2.0] :: Anaconda, Inc. on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from hdlConvertor import HdlConverter
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/poleguy/fpga-data/2023/vhdl_to_verilog/cenv/lib/python3.7/site-packages/hdlConvertor-2.3-py3.7-linux-x86_64.egg/hdlConvertor/__init__.py", line 1, in <module>
    from ._hdlConvertor import HdlConvertorPy as HdlConvertor, ParseException
ImportError: /home/poleguy/fpga-data/2023/vhdl_to_verilog/cenv/lib/python3.7/site-packages/hdlConvertor-2.3-py3.7-linux-x86_64.egg/hdlConvertor/_hdlConvertor.cpython-37m-x86_64-linux-gnu.so: undefined symbol: _ZNSt12experimental10filesystem2v17__cxx114path14_M_split_cmptsEv
>>>

poleguy, 2023-08-15:
Additional Info:
v2.0 seems to import fine
v2.1 produces a different "undefined symbol" error.

```
python
Python 3.7.16 (default, Jan 17 2023, 22:20:44) 
[GCC 11.2.0] :: Anaconda, Inc. on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from hdlConvertor import HdlConvertor
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/poleguy/fpga-data/2023/vhdl_to_verilog/cenv/lib/python3.7/site-packages/hdlConvertor-2.1-py3.7-linux-x86_64.egg/hdlConvertor/__init__.py", line 1, in <module>
    from ._hdlConvertor import HdlConvertorPy as HdlConvertor, ParseException
ImportError: /home/poleguy/fpga-data/2023/vhdl_to_verilog/cenv/lib/python3.7/site-packages/hdlConvertor-2.1-py3.7-linux-x86_64.egg/hdlConvertor/_hdlConvertor.cpython-37m-x86_64-linux-gnu.so: undefined symbol: _ZN12hdlConvertor4vhdl14BaseVhdlParserC2ERNS0_17VhdlCommentParserEb
>>> 

```

Nic30, 2023-08-21:
Hello,

which version of antlr4 C++ runtime and antrl4 are you using?
Can you try build from repo?
Did you tried a clean build? Is there a possibility that the build was interrupted and then re-executed ?

poleguy, 2023-08-21:
I've never used antlr4 myself. I installed it per the "Installation" section because you list it as a dependency. This is what I got:

````
$ apt list | grep antlr4

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

antlr4-doc/jammy,jammy 4.7.2-5 all
antlr4-maven-plugin/jammy,jammy 4.7.2-5 all
antlr4/jammy,jammy,now 4.7.2-5 all [installed]
golang-github-antlr-antlr4-dev/jammy,jammy 4.7.2+ds-2 all
libantlr4-runtime-dev/jammy,now 4.9+dfsg-1.1 amd64 [installed]
libantlr4-runtime-java/jammy,jammy,now 4.7.2-5 all [installed,automatic]
libantlr4-runtime4.9/jammy,now 4.9+dfsg-1.1 amd64 [installed,automatic]
python3-antlr4/jammy,jammy 4.9.1-1 all
````

I can't figure out how to try the build from the repo: I don't see a download link or clear instructions. Is it the pip3 install you mean? 

It seems the Installation entry in README.md is guiding me to building it myself.

I tried this and it failed:

````
$ pip3 install hdlConvertor
Defaulting to user installation because normal site-packages is not writeable
Collecting hdlConvertor
  Downloading hdlConvertor-2.3.tar.gz (665 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 665.7/665.7 KB 7.7 MB/s eta 0:00:00
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Requirement already satisfied: hdlConvertorAst>=1.0 in /usr/local/lib/python3.10/dist-packages (from hdlConvertor) (1.0)
Building wheels for collected packages: hdlConvertor
  Building wheel for hdlConvertor (pyproject.toml) ... done
  Created wheel for hdlConvertor: filename=hdlConvertor-2.3-cp310-cp310-linux_x86_64.whl size=1413968 sha256=d25d79f2914bab2e7afa1a47113a8f0e82332cdecb7e029bc59ebf80e3e7118f
  Stored in directory: /home/poleguy/.cache/pip/wheels/3f/c0/e7/83b5fc8863fdfd2d5704ba907782e7974d9157bf2df03dadb4
Successfully built hdlConvertor
Installing collected packages: hdlConvertor
Successfully installed hdlConvertor-2.3
$ python3
Python 3.10.12 (main, Jun 11 2023, 05:26:28) [GCC 11.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import hdlConvertor
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/poleguy/fpga-data/2023/dpsm_rx/hdlConvertor/hdlConvertor/__init__.py", line 1, in <module>
    from ._hdlConvertor import HdlConvertorPy as HdlConvertor, ParseException
ModuleNotFoundError: No module named 'hdlConvertor._hdlConvertor'
>>> from hdlConvertor import HdlConvertor
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/poleguy/fpga-data/2023/dpsm_rx/hdlConvertor/hdlConvertor/__init__.py", line 1, in <module>
    from ._hdlConvertor import HdlConvertorPy as HdlConvertor, ParseException
ModuleNotFoundError: No module named 'hdlConvertor._hdlConvertor'
>>> 
````
I did try a clean build. I deleted the repo and started fresh several times.

My system is on 12.3.0, and this does not work:
````
$ g++ --version
g++ (Ubuntu 12.3.0-1ubuntu1~22.04) 12.3.0
Copyright (C) 2022 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

$ gcc --version
gcc (Ubuntu 12.3.0-1ubuntu1~22.04) 12.3.0
Copyright (C) 2022 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
````

These are the commands I used to try to build:

````
rm -rf hdlConvertor/
git clone git@github.com:Nic30/hdlConvertor.git
cd hdlConvertor/
pip3 install --upgrade --force-reinstall --no-cache-dir git+https://github.com/Nic30/hdlConvertorAst.git
pip3 install -r requirements.txt
python3 setup.py install
````

poleguy, 2023-08-21:
Oh, wow, I think I may have discovered my problem. It does build correctly. But it doesn't run from the hdlConvertor directory. I needed to cd .. first before trying it.

I'm still not sure why just installing from python gave the same error.

But maybe it still didn't build right?
I couldn't find any examples of how I'm supposed to use this to convert vhdl to verilog, so I hacked something together from digging through the code/test code. The only example on the repo readme points to mybinder.org and doesn't show anything useful.

 I tried using it like this, and it gives an error:

````
Traceback (most recent call last):
  File "convert_hdl.py", line 21, in <module>
    verilog_code = vhdl_to_verilog(vhdl_code)
  File "convert_hdl.py", line 10, in vhdl_to_verilog
    ast, _ = hdlconv.parse([vhdl_code], Language.VHDL, ".")
TypeError: cannot unpack non-iterable HdlContext object
````

````
from hdlConvertor import HdlConvertor
from hdlConvertorAst.language import Language

def vhdl_to_verilog(vhdl_code):
    # Create a new instance of the HDLConvertor
    hdlconv = HdlConvertor()

    # Parse VHDL code and specify the input language
    ast, _ = hdlconv.parse([vhdl_code], Language.VHDL, ".")

    # Generate Verilog code from the parsed AST
    verilog_code = hdlconv.to[Language.VERILOG](ast)
    
    return verilog_code

# Example VHDL code
vhdl_code = "mux.vhd"

# Convert VHDL to Verilog
verilog_code = vhdl_to_verilog(vhdl_code)
print(verilog_code)
````

Using this file:
https://github.com/Nic30/hdlConvertor/blob/master/tests/vhdl/mux.vhd

Nic30, 2023-08-23:
> it doesn't run from the hdlConvertor directory
it is expected behavior, python imports local directory hdlConvertor as a library instead of build one which is installed.
You can just rename hdlConvertor directory or you can link builded libraries as described in "in place debugging" https://github.com/Nic30/hdlConvertor/blob/master/doc/dev_minitutorial.md

Nic30, 2024-01-14:
Closing this because it was resolved.

closed 2024-01-14 by Nic30