Nic30/hdlConvertor #190: Error in CMake based build

Siddhartha123, 2024-05-08:
Hi,
I am getting following error when trying to build hdlConvertor using setup.py

command - python3 setup.py install --prefix /tmp/python_install/ --build-type Debug -- -DANTLR_JAR_LOCATION=/home/sid/Downloads/antlr4-4.7.1-complete.jar

Below is the error log

```
[33/99] Building CXX object src/svConvertor/CMakeFiles/svConvertor_static.dir/attributeParser.cpp.o
FAILED: src/svConvertor/CMakeFiles/svConvertor_static.dir/attributeParser.cpp.o 
/usr/bin/c++  -I/usr/local/include/antlr4-runtime -I/home/sid/Documents/work/verilog/hdlConvertor/src/../include -Wall -pedantic -W -Wno-attributes -g -std=gnu++1z -fPIC -MD -MT src/svConvertor/CMakeFiles/svConvertor_static.dir/attributeParser.cpp.o -MF src/svConvertor/CMakeFiles/svConvertor_static.dir/attributeParser.cpp.o.d -o src/svConvertor/CMakeFiles/svConvertor_static.dir/attributeParser.cpp.o -c /home/sid/Documents/work/verilog/hdlConvertor/src/svConvertor/attributeParser.cpp
In file included from /home/sid/Documents/work/verilog/hdlConvertor/src/../include/hdlConvertor/svConvertor/attributeParser.h:5:0,
                 from /home/sid/Documents/work/verilog/hdlConvertor/src/svConvertor/attributeParser.cpp:1:
/home/sid/Documents/work/verilog/hdlConvertor/src/../include/hdlConvertor/svConvertor/sv2017Parser/sv2017Parser.h:304:43: error: ‘virtual const std::vector<std::__cxx11::basic_string<char> >& sv2017_antlr::sv2017Parser::getTokenNames() const’ marked ‘override’, but does not override
   virtual const std::vector<std::string>& getTokenNames() const override { return _tokenNames; }; // deprecated: use vocabulary instead.
                                           ^~~~~~~~~~~~~
[34/99] Building CXX object src/svConvertor/CMakeFiles/svConvertor_static.dir/commentParser.cpp.o
FAILED: src/svConvertor/CMakeFiles/svConvertor_static.dir/commentParser.cpp.o 
/usr/bin/c++  -I/usr/local/include/antlr4-runtime -I/home/sid/Documents/work/verilog/hdlConvertor/src/../include -Wall -pedantic -W -Wno-attributes -g -std=gnu++1z -fPIC -MD -MT src/svConvertor/CMakeFiles/svConvertor_static.dir/commentParser.cpp.o -MF src/svConvertor/CMakeFiles/svConvertor_static.dir/commentParser.cpp.o.d -o src/svConvertor/CMakeFiles/svConvertor_static.dir/commentParser.cpp.o -c /home/sid/Documents/work/verilog/hdlConvertor/src/svConvertor/commentParser.cpp
In file included from /home/sid/Documents/work/verilog/hdlConvertor/src/../include/hdlConvertor/svConvertor/commentParser.h:5:0,
                 from /home/sid/Documents/work/verilog/hdlConvertor/src/svConvertor/commentParser.cpp:1:
/home/sid/Documents/work/verilog/hdlConvertor/src/../include/hdlConvertor/svConvertor/sv2017Parser/sv2017Lexer.h:122:43: error: ‘virtual const std::vector<std::__cxx11::basic_string<char> >& sv2017_antlr::sv2017Lexer::getTokenNames() const’ marked ‘override’, but does not override
   virtual const std::vector<std::string>& getTokenNames() const override; // deprecated, use vocabulary instead
                                           ^~~~~~~~~~~~~
/home/sid/Documents/work/verilog/hdlConvertor/src/../include/hdlConvertor/svConvertor/sv2017Parser/sv2017Lexer.h:125:39: error: invalid covariant return type for ‘virtual const std::vector<short unsigned int> sv2017_antlr::sv2017Lexer::getSerializedATN() const’
   virtual const std::vector<uint16_t> getSerializedATN() const override;
                                       ^~~~~~~~~~~~~~~~
In file included from /usr/local/include/antlr4-runtime/Lexer.h:8:0,
                 from /usr/local/include/antlr4-runtime/antlr4-runtime.h:31,
                 from /home/sid/Documents/work/verilog/hdlConvertor/src/../include/hdlConvertor/svConvertor/commentParser.h:4,
                 from /home/sid/Documents/work/verilog/hdlConvertor/src/svConvertor/commentParser.cpp:1:
/usr/local/include/antlr4-runtime/Recognizer.h:58:36: error:   overriding ‘virtual antlr4::atn::SerializedATNView antlr4::Recognizer::getSerializedATN() const’
     virtual atn::SerializedATNView getSerializedATN() const {
                                    ^~~~~~~~~~~~~~~~
[35/99] Building CXX object src/svConvertor/CMakeFiles/svConvertor_static.dir/baseSvParser.cpp.o
FAILED: src/svConvertor/CMakeFiles/svConvertor_static.dir/baseSvParser.cpp.o 
/usr/bin/c++  -I/usr/local/include/antlr4-runtime -I/home/sid/Documents/work/verilog/hdlConvertor/src/../include -Wall -pedantic -W -Wno-attributes -g -std=gnu++1z -fPIC -MD -MT src/svConvertor/CMakeFiles/svConvertor_static.dir/baseSvParser.cpp.o -MF src/svConvertor/CMakeFiles/svConvertor_static.dir/baseSvParser.cpp.o.d -o src/svConvertor/CMakeFiles/svConvertor_static.dir/baseSvParser.cpp.o -c /home/sid/Documents/work/verilog/hdlConvertor/src/svConvertor/baseSvParser.cpp
In file included from /home/sid/Documents/work/verilog/hdlConvertor/src/../include/hdlConvertor/svConvertor/baseSvParser.h:5:0,
                 from /home/sid/Documents/work/verilog/hdlConvertor/src/svConvertor/baseSvParser.cpp:1:
/home/sid/Documents/work/verilog/hdlConvertor/src/../include/hdlConvertor/svConvertor/sv2017Parser/sv2017Parser.h:304:43: error: ‘virtual const std::vector<std::__cxx11::basic_string<char> >& sv2017_antlr::sv2017Parser::getTokenNames() const’ marked ‘override’, but does not override
   virtual const std::vector<std::string>& getTokenNames() const override { return _tokenNames; }; // deprecated: use vocabulary instead.
                                           ^~~~~~~~~~~~~
In file included from /home/sid/Documents/work/verilog/hdlConvertor/src/../include/hdlConvertor/svConvertor/commentParser.h:5:0,
                 from /home/sid/Documents/work/verilog/hdlConvertor/src/../include/hdlConvertor/svConvertor/baseSvParser.h:6,
                 from /home/sid/Documents/work/verilog/hdlConvertor/src/svConvertor/baseSvParser.cpp:1:
/home/sid/Documents/work/verilog/hdlConvertor/src/../include/hdlConvertor/svConvertor/sv2017Parser/sv2017Lexer.h:122:43: error: ‘virtual const std::vector<std::__cxx11::basic_string<char> >& sv2017_antlr::sv2017Lexer::getTokenNames() const’ marked ‘override’, but does not override
   virtual const std::vector<std::string>& getTokenNames() const override; // deprecated, use vocabulary instead
                                           ^~~~~~~~~~~~~
/home/sid/Documents/work/verilog/hdlConvertor/src/../include/hdlConvertor/svConvertor/sv2017Parser/sv2017Lexer.h:125:39: error: invalid covariant return type for ‘virtual const std::vector<short unsigned int> sv2017_antlr::sv2017Lexer::getSerializedATN() const’
   virtual const std::vector<uint16_t> getSerializedATN() const override;
                                       ^~~~~~~~~~~~~~~~
In file included from /usr/local/include/antlr4-runtime/Lexer.h:8:0,
                 from /usr/local/include/antlr4-runtime/antlr4-runtime.h:31,
                 from /home/sid/Documents/work/verilog/hdlConvertor/src/../include/hdlConvertor/svConvertor/sv2017Parser/sv2017Parser.h:7,
                 from /home/sid/Documents/work/verilog/hdlConvertor/src/../include/hdlConvertor/svConvertor/baseSvParser.h:5,
                 from /home/sid/Documents/work/verilog/hdlConvertor/src/svConvertor/baseSvParser.cpp:1:
/usr/local/include/antlr4-runtime/Recognizer.h:58:36: error:   overriding ‘virtual antlr4::atn::SerializedATNView antlr4::Recognizer::getSerializedATN() const’
     virtual atn::SerializedATNView getSerializedATN() const {
                                    ^~~~~~~~~~~~~~~~
[36/99] Building CXX object src/CMakeFiles/hdlConvertor_cpp_static.dir/hdlConvertor.cpp.o
FAILED: src/CMakeFiles/hdlConvertor_cpp_static.dir/hdlConvertor.cpp.o 
```

Siddhartha123, 2024-05-08:
Do I need to use any particular version of CXX toolchain? Doc mentions anything above gcc-9 should work.
Alternatively it would be really helpful if someone can provide pre-built wheel/binary which I can directly install

Thanks

Nic30, 2024-05-21:
Hello,
I think you are building against the wrong version of antrl4 (cpp runtime).
You can try mesonbuild branch.

see https://github.com/Nic30/hdlConvertor/issues/187#issuecomment-2122813878

Nic30, 2025-06-10:
cmake replaced with meson, closing this issue.

closed 2025-06-10 by Nic30