Nic30/hdlConvertor #193: meson build fails, unable to find <hdlConvertor/verilogPreproc/verilogPreprocParser/verilogPreprocLexer.h>

crsumiran, 2024-07-22:
I have checked out the meson build branch and ran 
meson setup builddir
meson compile -C builddir

I am getting this error. what am I missing. I am unabl 

FAILED: src/libhdlConvertor_cpp_static.a.p/hdlConvertor.cpp.o 
c++ -Isrc/libhdlConvertor_cpp_static.a.p -Isrc -I../src -Iinclude -I../include -I../subprojects/antlr4-runtime/runtime/Cpp/runtime/src -I../subprojects/antlr4-runtime/runtime/Cpp/runtime/src/atn -I../subprojects/antlr4-runtime/runtime/Cpp/runtime/src/dfa -I../subprojects/antlr4-runtime/runtime/Cpp/runtime/src/internal -I../subprojects/antlr4-runtime/runtime/Cpp/runtime/src/misc -I../subprojects/antlr4-runtime/runtime/Cpp/runtime/src/support -I../subprojects/antlr4-runtime/runtime/Cpp/runtime/src/tree -I../subprojects/antlr4-runtime/runtime/Cpp/runtime/src/tree/pattern -I../subprojects/antlr4-runtime/runtime/Cpp/runtime/src/tree/xpath -Isubprojects/antlr4-runtime/__CMake_build -I../subprojects/antlr4-runtime/__CMake_build -Isubprojects/antlr4-runtime -I../subprojects/antlr4-runtime -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -std=c++17 -O0 -g -fPIC -Wall -pedantic -W -Wno-attributes -MD -MQ src/libhdlConvertor_cpp_static.a.p/hdlConvertor.cpp.o -MF src/libhdlConvertor_cpp_static.a.p/hdlConvertor.cpp.o.d -o src/libhdlConvertor_cpp_static.a.p/hdlConvertor.cpp.o -c ../src/hdlConvertor.cpp
In file included from ../include/hdlConvertor/hdlConvertor.h:14,
                 from ../src/hdlConvertor.cpp:1:
../include/hdlConvertor/verilogPreproc/verilogPreproc.h:13:10: fatal error: hdlConvertor/verilogPreproc/verilogPreprocParser/verilogPreprocLexer.h: No such file or directory
   13 | #include <hdlConvertor/verilogPreproc/verilogPreprocParser/verilogPreprocLexer.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

Thomasb81, 2024-07-22:
On your system the file `hdlConvertor/verilogPreproc/verilogPreprocParser/verilogPreprocLexer.h` is missing.

The compilation flow is supposed to generate this file with antlr4 tool from grammar file located in grammar directory.
The antlr4 tool is suppose to be download from [antlr-4.13.1-complete.jar](https://www.antlr.org/download/antlr-4.13.1-complete.jar) by the build system.
Then it require java to execute the tool.

Without having more details on your system and environment, nobody can't guess what is the root issue at your end.

crsumiran, 2024-07-22:
If I install antlr and run the below steps, will it be sufficient? 
meson setup builddir
meson compile -C builddir

Thomasb81, 2024-07-22:
What do you mean by sufficient ?
It depends what you expect to do.
```
meson setup builddir
meson compile -C builddir
```
I think, it will limit the build to the C++ code... 

You can found ideas here https://github.com/Nic30/hdlConvertor/blob/mesonbuild/.circleci/config.yml

I personally work in a python virtualenv,
install meson-python 
```
pip install meson-python
```
Then build the library inplace, and with debug symbol with:
```
pip install --no-build-isolation --editable . -Csetup-args=-Dbuildtype=debug -Cbuild-dir=build-dbg
```

crsumiran, 2024-07-23:
@Thomasb81 thank you so much. I followed the steps in https://github.com/Nic30/hdlConvertor/blob/mesonbuild/.circleci/config.yml
This worked.

closed 2024-07-23 by crsumiran