Nic30/hdlConvertor #199: Does hdlconvertor not support function keyword?

JiinJie, 2025-03-19:
### I found the problem when I used "hdlconvertor"
_The python code is as follows_

> 
> from hdlConvertor import HdlConvertor
> from hdlConvertorAst.language import Language
> 
> parser = HdlConvertor()
> 
> converter = HdlConvertor()
> ast = converter.parse(filenames=["vhd_function.vhd"], language=Language.VHDL_2008, encoding='utf-8', incdirs=["include_dir"])
> 
> print(ast)
> 

_The vhdl code is as follows_

> library IEEE;
> use IEEE.STD_LOGIC_1164.ALL;
> 
> function AND_FUNC(A, B : STD_LOGIC) return STD_LOGIC is
> begin
>     return A and B;
> end function AND_FUNC;
> 
> entity Example is
>     Port ( A : in STD_LOGIC;
>            B : in STD_LOGIC;
>            C : out STD_LOGIC);
> end Example;
> 
> architecture Behavioral of Example is
> begin
>     C <= AND_FUNC(A, B);
> end Behavioral;

_The following error message is displayed_

> Traceback (most recent call last):
>   File "D:\PyCharm_projects\test_function_AST.py", line 42, in <module>
>     ast = converter.parse(filenames=["vhd_function.vhd"], language=Language.VHDL_2008, encoding='utf-8', incdirs=["include_dir"])
>   File "_hdlConvertor.pyx", line 173, in hdlConvertor._hdlConvertor.HdlConvertorPy.parse
>   File "_hdlConvertor.pyx", line 49, in hdlConvertor._hdlConvertor.raise_cpp_py_error
> hdlConvertor._hdlConvertor.ParseException: 
> vhd_function.vhd:5:0:SyntaxError:mismatched input 'function' expecting {KW_CONTEXT, KW_ENTITY, KW_PACKAGE, KW_CONFIGURATION, KW_ARCHITECTURE}

fisherdog1, 2025-03-30:
You cannot declare a function there because it is not a primary or secondary unit (13.1 in the VHDL 2008 LRM). You can declare subprograms in basically any _declarative_ region (e.g. entity, package, architecture, process)

Nic30, 2025-06-10:
As fisherdog1 said the syntax is incorrect, that is why it does not work.
You already did put it inside of some wrapper or you started parsing on rule for function.
Closing this issue as it was long time ago.